xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #3: TL-WDR4300 in 2.4GHz mode

mzakharo, 2017-07-10:
Hello,

The  OpenWRT [Guide](http://pdcc.ntu.edu.sg/wands/Atheros/document/Atheros-CSI-Tool-User-Guide(OpenWrt).pdf) on the website mentions that the code has been tested with TL-WDR4300, which uses AR9580. However, I can only seem to be able to use it in 5Ghz band. Are there any tips to get AR9580 to operate in 2.4GHz band in WDR4300 platform?

xieyaxiongfly, 2017-07-11:
Are you using WDR4300?  WDR4300 has two WI-Fi cards, one is 9580, which
works on 5G. Another one is AR9340, which has only 2 antennas and works on
2.4GHz.

These two WI-FI cards are capable of extracting CSI. So you should work
without any problem at 2.4GHz. But you are using AR9340 instead of AR9580
if at 2.4GHz.

I think we are not able to change the setting and use AR9580 at 2.4GHz, or
at least I don't know such a method to achieve that.

Hope this answers your question.

mzakharo, 2017-07-11:
Thank you,  I just thought that AR9580 was a dual-band radio transceiver, but it seems like EEPROM has 2.4GHz disabled.

closed 2017-07-11 by mzakharo

xieyaxiongfly, 2017-07-11:
AR9580 is a dual-band radio transceiver. But WDR4300 only use it at 5GHz.