xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #6: Router bricks after flashing the image

eroniki, 2017-10-24:
As the title goes, after compiling the repo and flashing the open-wrt images to the router, it becomes unresponsive. (open-wrt images provided in the open-wrt.org works well)

I repeated the process couple of times and no luck. The images are compiled with a Ubuntu 14.04 machine (kernel: 4.4.0.97). I have no missing dependency or problem in the process of make. 

I am wondering if someone else can compile the repo with a different computer and post the results. 

Thanks!

xieyaxiongfly, 2017-10-24:
I can send you some binary image if you are interested.

eroniki, 2017-10-24:
> I can send you some binary image if you are interested.

@xieyaxiongfly I would appreciate it!

xieyaxiongfly, 2017-10-24:
Can you tell me what router you are using?

eroniki, 2017-10-24:
The same in the documentation. TP-Link 4300.

xieyaxiongfly, 2017-10-24:
The binary file is here.
https://www.dropbox.com/s/vgu5lk85s2rpvzl/openwrt_wdr4300_img.tar.gz?dl=0

eroniki, 2017-10-24:
Your image seems to work. I will try to identify what the problem is resulting from. Please keep this issue until I post the conclusion.

lazynihilist, 2018-04-20:
Can you send the binary for TL-WR740N v4.27 and TL-WR841N v9.1 ?

I have bricked both of my routers and I badly need them to capture CSI.

nilspo, 2018-06-14:
@xieyaxiongfly Does your posted binary file for WDR4300 support injector mode?
@eroniki Did you identify your problem? I had the same problem and I have no idea what I did wrong. The uploaded image worked.

bharath-krishnaswamy, 2019-02-10:
The same thing is happening with my TP link archer C60 v2.0 router

eroniki, 2019-02-13:
> @xieyaxiongfly Does your posted binary file for WDR4300 support injector mode?
> @eroniki Did you identify your problem? I had the same problem and I have no idea what I did wrong. The uploaded image worked.

Same here. For some reason my own binaries bricked the router. I was able to recover with @xieyaxiongfly's

Alpha2Cool, 2019-10-15:
Version problem? v1.0? v2.0?

Geosearchef, 2019-10-15:
There is only v1.0-v1.7

sarashoouri, 2024-06-12:
can I have the binary for WNDR4300 (Netgear)