xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #11: Not working at the injector mode on Openwrt system

ClarenceZSK, 2018-05-31:
Dear Dr. Xie,

I have built your CSI tool and installed it on my router Netgear WNDR3700v4. It shows that the CSI works well as your tutorial suggests. But I do not know what wrong with the injector mode.

I have set all devices (one router and one Atheros 9580 card installed on my Ubuntu machine) to work on monitor mode, following your tutorial to modify all devices' MAC addresses until the command "$ iw dev mon0 set freq 2437".

I would like to use the router to inject packets to my machine. So I run the recvCSI app on the machine first to wait for the injection packets from the router. Then I run the command  "$ injector -i mon0 -c 6HT20 -m 0 -n 100 -d 1000" on the OpenWRT. Nothing happened. There is no command delay, no log, and no packet received by my machine. 

Is there something wrong for my setting? Looking forward to your reply. Thank you very much.

ClarenceZSK, 2018-06-06:
I have fixed the issue. The Makefile of lorcon package has a bug that "PKG_SOURCE_VERSION:=80be82a9bee088dc9ebd42fd61366b7abe182331" should be "PKG_SOURCE_VERSION:=3c2c1d398f6caaca0f524d397003212932b55ec3". 

But I have another issue that I cannot receive the CSI when transmitting 40MHz packets by setting the option -b of injector to be 1. It works well when transmitting 20MHz packets. Could you please shed some light on that? @xieyaxiongfly  Thank you very much.

ClarenceZSK, 2018-06-08:
@xieyaxiongfly Is there any method to do antenna selection for the injector? The injector only chooses one antenna to get the send packet so that I receive a 3*1*56 CSI matrix. I would like to get a 3*3*56 CSI matrix in the injection-monitor mode.

xieyaxiongfly, 2018-06-11:
For some unknown reasons, the Lorcon software cannot inject MIMO packets. I
can fix that problem and provide some interfaces for you to select the
antennas.

I am back on coding these days, I expect this to be done within 1 or 2
months.

xieyaxiongfly, 2018-06-11:
> HI Yaxiong,
>
> Thank you for developing this tool and making it available for us.
>
> @ClarenceZSK <https://github.com/ClarenceZSK> thanks for the correction.
> I am able to get the CSI via injector mode as well. Even I am seeing the
> same issue. When I analyse on WireShark, I see that when we set -b to 1
> (40MHz), there are no packets detected on WireShark. So maybe they are not
> being injected from router side.

The bandwidth problem is not that easy to solve. I didn't figure out how to
inject packet with 40MHz. But you can transmit 40MHz packet in AP-Client
mode.

Alpha2Cool, 2019-10-22:
Compile too slowly，几乎没有动静，是因为国内需要翻墙吗，lol
[image]

Alpha2Cool, 2019-10-23:
> I have fixed the issue. The Makefile of lorcon package has a bug that "PKG_SOURCE_VERSION:=80be82a9bee088dc9ebd42fd61366b7abe182331" should be [...]

I can't finish compiling firmware when I fix as you say, I can't inject packets, help....