xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #13: Can you list some other usable wifi routers besides  TL-WDR4300?

PigSea, 2018-08-14:
Hello, since OpenWRT Guide employs TL-WDR4300 which is not in production now,  have you realized it on other wifi routers? I would greatly appreciate if you could give me some advices about other available wifi routers I can choose for tool installation. Thank you.

xieyaxiongfly, 2018-08-14:
Check this page
http://wands.sg/AtherosCSI/Hardware.html

PigSea, 2018-08-14:
ok, I see, thanks a lot.

bharath-krishnaswamy, 2018-10-26:
Can i use TPLINK TL-WR740N for creating the CSI available version? I've checked the list, but there is no firmware for the mentioned router. How to create the firmware?

luxiangx, 2018-11-28:
@xieyaxiongfly 

I can not open the URL. Maybe it changes.

shulinyang, 2019-02-18:
Can you list some other usable wifi routers besides TL-WDR4300?

wuhaozheng2, 2019-07-10:
@shulinyang Have you tried any routers?
I'm looking for a available routers list but the link above was 404.