xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #14: Using this tool on Atheros AR9280

akhila-s-rao, 2018-08-22:
Hi, 

I have been desperately trying to get this tool to work on my PC engines AP fitted with now AR9280 wifi NIC. I first started with QCA9882 which did not work and I hence downgraded to AR9280 because this is what my vendor had. Could I please get some information about whether this wifi NIC will be successful in getting CSI. Is it a maybe or a definite yes/no ? I see that it is listed on the supported hardware page but it says that noone has tested it yet. I ask because I cannot check myself at this point since I do not have a sender that has the Atheros chipset. My laptop has an Intel 8260 wifi NIC.

xieyaxiongfly, 2018-08-22:
> Missed a thank you at the end : )

I think the answer is 90% no. AR9280 cannot work with our tool.

sedkiyounis, 2020-02-21:
I am just wondering  if the tool will work with AR9285