xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #18: Unstable CSI amplitude

hanstseng123, 2018-08-30:
I use TP-LINK wdr4300 as AP and client which are equipped 3 antennas(although they can only use 2 antennas ) to measure CSI.
I fixed both the position of AP and client and measured CSI amplitude at three different time for testing the temporal stability.
I have make sure that there is no object changing when I was sending data.
There is the result which have 500 samples in each condition :   
 [image]

It looks unreasonable.
what may cause this problem?
Did anyone meet this problem before?

xieyaxiongfly, 2018-08-30:
No, this is totally reasonable.

Please take a look at this slides.
http://www.cs.princeton.edu/~yaxiongx/doc/CSI_compare.pptx

hanstseng123, 2018-10-17:
> @hanstseng123 Hi, could you please tell me how did you normalize the CSI matrix data? Or you just use the extracted data to measure CSI amplitude?

I transform the CSI amplitude with matlab command db( ).

bigtreeyanger, 2019-06-04:
> No, this is totally reasonable. Please take a look at this slides. http://www.cs.princeton.edu/~yaxiongx/doc/CSI_compare.pptx
> […](#)

@xieyaxiongfly Hi, I also observed the unstable amplitudes. I found the link to the slides is no longer valid, I am wondering where I can find them now and I intend to use them as a reference. Tks!

chili6f, 2019-07-31:
> > No, this is totally reasonable. Please take a look at this slides. http://www.cs.princeton.edu/~yaxiongx/doc/CSI_compare.pptx
> 
> @xieyaxiongfly Hi, I also observed the unstable amplitudes. I found the link to the slides is no longer valid, I am wondering where I can find them now and I intend to use them as a reference. Tks!

Hello, if you managed to find those slides, could you send me a link? I am observing similar unstable amplitudes. 

Help me, @bigtreeyanger , you're my only hope.

[image]

bigtreeyanger, 2019-08-02:
Hi @chili6f,  sorry I haven't found the slides, and my observations are similar to yours.

yujianyuanhaha, 2021-03-06:
@hanstseng123 @bigtreeyanger it simply channel fading...