xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #23: Archer C7(V2 and V4)  5Ghz csi

ahj3460, 2018-10-24:
Hello.

Extracting 2.4Ghz CSI with Archer C7 (V2 and V4) has been successful.

But I want to extract 5GHz CSI. I don't know what method to use.

I would appreciate your help.

shulinyang, 2019-04-13:
如何获得114个子载波呢？

shulinyang, 2019-04-15:
能发一些路由器购买链接吗@kdkalvik

kdkalvik, 2019-04-15:
This works for me https://www.amazon.com/TP-Link-Wireless-Wi-Fi-Router-TL-WDR4300/dp/B0088CJT4U

shulinyang, 2019-04-15:
您知道在ubuntu模式如何获得114个子载波呢？如何改成5ghz?感谢

shulinyang, 2019-04-15:
您知道在ubuntu模式如何获得114个子载波呢？如何改成5ghz?感谢@kdkalvik

kdkalvik, 2019-04-15:
follow the instructions here for setting MCS rate https://github.com/xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src/issues/27#issuecomment-449050917, 
use luci to set the AP to 5 GHz

shulinyang, 2019-04-15:
我使用的是Ubuntu的版本,没有使用路由器和openwrt，那么在ubuntu版本下怎样操作呢？@kdkalvik

kdkalvik, 2019-04-15:
You can set the MCS rate with the same command, just make sure you select the right phy device. And refer to this https://medium.com/@renaudcerrato/how-to-build-your-own-wireless-router-from-scratch-part-3-d54eecce157f for setting the AP in 5GHz. you might have to modify the patch based on your kernel version.

shulinyang, 2019-04-15:
谢谢，已经处理好了，您能分享一下处理114个子载波数据的代码吗？1418373708ysl@gmail.com。万分感谢@kdkalvik

kdkalvik, 2019-04-15:
The thing with 114 subcarriers is that there is too much CSI data so the driver splits it into 2 packets (usually the csi data comes in a single packet). The hack does not properly handle this case. The hack stores the first packet with csi in the payload and second packet in csi buffer. When we decode the payload buffer alone as a csi packet we endup getting 0x3x114 data. You can probe this [part](https://github.com/xieyaxiongfly/Atheros-CSI-Tool/blob/2288dcf91da81365c388b23a0f6cc113e41dde2b/drivers/net/wireless/ath/ath9k/ar9003_mac.c#L611)  of the ar9003_mac.c to get a better picture of the problem. But long story short if you want to recover 3x3x114 you have to know which packets were recorded as 0x3x114 and combine the payload and csi buffers and then decode the combined buffers to get the total 3x3x114 csi data. I tried o fix the driver but I couldn't figure out a clean solution. You can give it a shot if you like and if you don't care about the payload of the packet and just want the csi, you can just update the provided matlab script to combine the buffers.

shulinyang, 2019-04-16:
出现一个问题，Ubuntu模式下，天线设置为3*1，结果收集的数据是3*2*114，您能解决这个问题吗？万分感谢@kdkalvik

zwlovefish, 2019-05-17:
@shulinyang 你好，请问你是如何获得114个子载波的？我按照#27那个将那个文件写入78，但还是3*2*56个子载波。。。

closed 2019-07-02 by ahj3460