xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #25: Getting 56 subcarriers in 40 MHz

kdkalvik, 2018-11-27:
Hi,
I have a WDR4300 setup at 5GHz 40 MHz and a desktop client connected to the router. I used wavemon to verify that the connection is indeed at 40MHz but the recv_csi tool is giving me only 56 sub carriers. I tried using another WDR4300 router as the client at 40MHz and still got 56 sub carriers, I also setup a desktop as the AP at 5GHz 40MHz with hostapd and used another desktop as the client and got only 56 sub carriers. I have the AR9380 NIC on the desktops. Did anyone manage to get all 114 sub carriers?

Djydo, 2018-11-28:
Hi @kdkalvik,

Yes, it is possible to get CSI on 114 subcarriers at 5 GHz. Set AP (4300 router) to 40 MHz using luci or  OpenWrt GUI **before** associating the client to the AP.  The AP is running OpenWrt version of atherosCSI and client is using ubuntu 4.1+ kernel.

kdkalvik, 2018-11-28:
I tried that but for some reason I keep getting only 56 sub carriers. Got any suggestions?

kdkalvik, 2018-11-28:
Yeah, I used the web interface to check 40MHz and it was indeed using 40MHz. Even when I used desktops on both AP and client, I was ble to verify that I have 409MHz with wavemon. Not sure what the issue is.

kdkalvik, 2018-12-16:
Figured it out, I was setting the MCS rate to 16 on the transmitter by updating the rate_index_table by echoing 18 into it. But it is supposed to be 78, to set it to MCS 16 at 40MHz.

closed 2018-12-16 by kdkalvik