xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #27: Getting 0x3x114 CSI values

kdkalvik, 2018-12-16:
I have a WDR4300 router setup with openwrt as the transmitter and a desktop with the ar9590 NIC for the receiver. I forced MCS 16 at 40 MHz (5Ghz) by updating the rate_index_table with value 78. But, when I transmit packets, I am getting 0 transmitting antennas. I tried setting MCS 15 at 40 MHz with 77 in the rate_index_table file and I am getting 2x3x114 CSI values. Not sure why it won't  transmit with all antennas when using MCS 16 at 40 MHz. I tried using different routers for transmitting and different desktops with different NICs and got the same issue. Lastly, I am receiving 3x3x56 CSI values when using MCS 16 at 20 MHz.

thu-zxs, 2018-12-20:
same issue here. When adjusting to a higher rate, I keep getting 3x0x114 CSI value. Not sure what happens.

btw, what is the `rate_index_table` file you mentioned?

kdkalvik, 2018-12-20:
When you echo a number into the fixed_rate_idx file you can adjust the MCS rate. 18 gives you MCS 16 at 20MHz and 78 is MCS 16 at 40MHz. you can try playing around with the value to change the Guard interval, MCS rate and channel width.

echo 78 >> /sys/kernel/debug/ieee80211/phy1/rc/fixed_rate_idx

kdkalvik, 2018-12-20:
Update: It looks like when ever I send packets with any length of payload at MCS 16 or higher at 40MHz, the recv_csi tool receives a fixed payload length of 1924. Not sure why though.

thu-zxs, 2018-12-21:
Yeah, right. When echoing `78` into that files, packets with payload length of 1924 keeps occurring, with CSI value being 3x0x114. When value being `77`, payload length is 1040, and CSI value is 3x2x114.

MCS 16 corresponds to a 3 streams transmission, which may cause problems according to the author:
https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/issues/22#issuecomment-287285288

I've sent email to @xieyaxiongfly  with no response. Hope anyone could help, really need 3x3x114 CSI.

jinzhang-unsw, 2018-12-25:
Any updates? I use ubuntu version and any higher than mcs 16 will mostly lose CSI with ampty 3*0*114 and few 3*3*56 ones.

thu-zxs, 2018-12-26:
Nothing. Got stuck.

jinzhang-unsw, 2018-12-26:
I print out the buffer in ```csi_fun.c``` in the receiver, the csi_len variable is wrong, and ```buf_addr``` mostly is 0 when transmit in any rates higher than 0x90 (MCS 16). Everything work normally if transmit in rates lower than 0x8f (MCS 15). Potentially the transmitter has bug like bit operation can cause this problem.

shulinyang, 2019-04-10:
ubuntu版本下如何设置成5GHz，如何获得114个子载波？@jinzhang-unsw

shulinyang, 2019-04-10:
ubuntu版本下如何设置成5GHz，如何获得114个子载波？@thu-zxs

thu-zxs, 2019-04-15:
> ubuntu版本下如何设置成5GHz，如何获得114个子载波？@thu-zxs

1. AP working on 5GHz: Google about the **hostapd configurations**.
2. search on the issues for how to change the transmission rate on ubuntu.

shulinyang, 2019-04-15:
能分享一下处理114个子载波的代码吗@thu-zxs

thu-zxs, 2019-04-15:
> 能分享一下处理114个子载波的代码吗@thu-zxs

只要ap设置成40MHz，设置好传输速率就可以获得114个子载波呀。

shulinyang, 2019-04-16:
已经可以拿到114个子载波，您有源码对114子载波进行处理吗，如一些过滤器？@thu-zxs

shulinyang, 2019-04-17:
如何调整速率@thu-zxs

shulinyang, 2019-04-17:
如何调整速率@jinzhang-unsw

zwlovefish, 2019-05-30:
@shulinyang 你好，请问你是怎么拿到114个子载波的啊？可否加一波好友哇

shulinyang, 2019-06-24:
> @shulinyang你好，请问你是怎么拿到114个子载波的啊？可否加一波好友哇

18255445041微信

shulinyang, 2019-06-24:
18255445041微信

shulinyang, 2019-06-25:
1418373708or微信18255445041

shulinyang, 2019-07-02:
我现在可以拿到114个子载波，但是数据中每次都会存在3*2*114而且会存在56的子载波，请问您遇到这种情况吗？如何修改呢@thu-zxs

mzakharo, 2019-11-10:
1924 payload is a bug in Linux kernel CSI driver. 1924 is not the payload, it is a partial CSI buffer. See my comment: 
https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/issues/22#issuecomment-552143069