xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #30: Installed Firmware on Archer C60 V2; not able to access openwrt

army-mehak, 2019-04-10:
I flashed the router Archer C60 v2 firmware using the web browser. After the update, I wasn't able to connect it through using http://192.168.1.1. So through "PUTTY", I established an SSH connection on 192.168.1.1, where Im trying to download "opkg update" and "opkg install luci".
I have received an error on Putty "opkg download error; failed to establish connection". Any suggestion?

closed 2019-04-12 by army-mehak

kr689, 2023-01-17:
How did you manage to resolve thist? I also installed the firmware in TL-WDR4300 v1. After what seemed to b a successful flash, I cannot seem to access 192.168.1.1.

Geosearchef, 2023-01-18:
Did you use an image containing LUCI? Can you reach ping the device? Can you ssh into the device?

kr689, 2023-01-18:
Update : So following this: https://openwrt.org/docs/guide-quick-start/ssh_connect_to_the_internet_and_install_luci#:~:text=Connect%20the%20PC%20to%20the,be%20able%20to%20install%20LuCI.

I was able to connect to the internet and was able to ssh into the device using the new IP configured. 
I now have a new problem when I try to install luci:
[image]
Any ideas what the issue is now and how I can get around that?

Geosearchef, 2023-01-18:
About getting the software, as you figured out, you need to connect the router to the internet first using the method you linked.

Alternatively, you can also download the software you need from an openwrt web page after selecting the correct architecture and copy the package file to the router using SFTP, and then install via opkg from file.

About the bricked router, are you sure you used the right (factory VS sysupgrade) image for the right device?
You could try to figure out the routers IP addr using Wireshark.

About recovering, a functional ethernet connection during operation doesn't matter. The bootloader will create a short connection during boot and immediately close that again. You should be able to see that connection attempt to your TFTP server using Wireshark.
Usually this then allows you to recover the device, as the bootloader is unaffected by you flashing a new image.

About installing LUCI, there seems to be a dependency conflict. Maybe you're using an old version of the packet? Did you update the package indices? You can also try manually getting the package and copying+installing from file.
Maybe also ask on an openwrt forum, they know more about this than here.

kr689, 2023-01-18:
So I had a relatively new version of OpenWrt running on the router and wanted to go to the precompiled CSI tool available version. I used the sysupgrade.bin file. The image check failed but I went ahead and forced the upgrade because it said that if downgrading or going to vendor firmware, the image check will fail and you will need to force the upgrade. it didn't work though I guess.

Right, that is good news. I've gone through the process using SolarWinds so I'll try that again.

Yes, that's where I'll head. 

Cheers for the help. I'll comment in how I solve this.