xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #32: Receive CSI tool

army-mehak, 2019-04-12:
**Thank you for this tool :) I am trying to use it for one of my projects.**
_I have achieved the following:_
My first Dell laptop (lets call it **"A"**) is connected to a router (TP-LINK Archer C60 v2) installed with OpenWRT system, which can be accessed via Luci on _http://192.168.1.1_. I have set up an access point through the Luci webpage as per your "Collect CSI" instructions.
My other dell laptop (lets call it **"B"**) contains Atheros 9462 Wifi card.  I have followed the tutorial for setting up Atheros tool on Ubuntu.  I even installed the Atheros CSI Tool User-App on this.
I connected laptop "B" to the wifi which I configured through laptop A Luci browser. As per my knowledge, they both are in the same network.
I am not able to run receive CSI command.

I think its due to one of these reasons
(1) When I set up the ubuntu version of CSI tool, the command "uname -r" returned 4.4.0-142 and not 4.1.10. How do I establish this?
(2) I downloaded Atheros CSI Tool User App on laptop A also and was trying to send packets. Is this how I do it? I download this GitHub repo in both?

Thank you for your help in advance 💯

Geosearchef, 2019-06-05:
(1) If you followed the tutorial for installing the built kernel you should be able to boot it by selecting sth along the linse of "Advanced Options" in GRUB.
(2) You need to send the packets on one of the devices running the "CSI kernel" to the other device with the csi tool.

salmanpolito, 2020-01-09:
Hey.. I am trying to install CSI tool on archer C7-V2 router. I have applied the patch and have checked the code lines one by one. it can alse grabe ar9003_csi when logged in to the router, but when I try to send packets to the same other router. The other router cannot receive any packet (although it is in listening mode). Can you please help what shall I do. It will be grate if you could share your subupgrade bin file. many thanks