xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #35: Modifying the CSI-Tool in the OpenWRT build

Geosearchef, 2019-06-12:
I have a setup for capturing CSI consisting out of a Laptop (14.04, AR9462) and multiple routers (Archer c7v5, QCA9563) running OpenWRT. I have associated them, the laptop being the AP (using hostapd, WPA2).

Ubuntu is running the right kernel, is able to receive CSI information from the router. I'm unable though to receive CSI data in the Ubuntu->OpenWRT direction. Both devices have the csi kernel module loaded, on OpenWRT the wlan1 interfaces is also listed as using the ath9k driver while the dmesg log contains `device class created correctly` and logs opening and closing `/dev/CSI_dev`.

To debug this I'm currently trying to modify the CSI tool itself (https://github.com/xieyaxiongfly/Atheros-CSI-Tool/) but can't figure out how to apply the changed code to the openWRT build. Where is this repo included in the build?
Also, do you have any idea why I can't receive CSI data / how I can debug this.

Geosearchef, 2019-06-14:
Am I correct in assuming it's in the patches/files in `package/kernel/mac80211/patches/556-ath9k_CSI_*****.patch`  or `package/kernel/mac80211/files`?

b00l3an, 2019-07-19:
Did you manage to resolve the issue and get QCA9563 to transmit / receive ?

Geosearchef, 2019-07-19:
When transmitting with QCA9563, everything seems to be fine, the not sounding flag is not set -> therefore sounding, I can receive it fine on an Ubuntu laptop with the AR9462 (AR5B22), but on the QCA9563 I don't get CSI.

I'm encountering one issue with the CRCErr not being detected which the tool uses to check for being on the last subframe (https://github.com/xieyaxiongfly/Atheros-CSI-Tool/issues/40). When disabling this check, I still don't get any CSI as the device is not uploading the data to the driver (which would be indicated via the hw_data_upload, hw_data_upload_valid and hw_data_upload_type flags, take a look at the MAC, RX descriptor section of e.g. the AR9344 datasheet). Those flags are set when a sounding packet is received, the CSI has been calculated and the hardware is about the inform the software. The ath9k actually doesn't relay this info back to the beamformer (station it came from) as a lot of manufacturers didn't opt in to implementing this into their hardware till 802.11ac. I recommend reading the IEEE 802.11n standard, section 19/20 I think about sounding PPDUs and Transmit Beamforming (TxBF, the technique used here to get the CSI).

I'm guessing this might be due to a newer chip (the AR9344 seems to list TxBF as a feature, I haven't seen it on the QCA9563 so far, e.g. on wikidevi) so I'm currently trying to get access to a device/router with a QCA9558 (e.g. Archer C7v1-3, although v1 doesn't support OpenWRT, v1-3 uses internal 2.4GHz antennas), a AR9380 (connected via PCIe) or a AR9580 (both of which the CSI tool was tested for) chip.

@ xieyaxiongfly's newest paper e.g. uses the Compex WPJ558 with the AR9340, AR9580 and QCA9558 chips I think. (https://arxiv.org/pdf/1812.03103.pdf)

Here's my list of routers I think could be of interest which are pretty limited as I need 3 external antennas. You can identify devices easily by going to wikidevi's atheros/qualcomm atheros pages and clicking the devices link there on any chip to see routers with that chip.

[image]

You should be able to find them for resale on e.g. ebay pretty easily.

b00l3an, 2019-07-20:
Thanks Geo. I believe I have encountered the same issue, I have 3 routers all with the QCA9563 (WR1043N Vers 5 & Archer C7 1750 Vers 5), it was the closest to QCA9558 I could source easily, and couldn't get the sounding flag to send. So its glad to have an explanation.

Thank you for the list of hardware though, very helpful

Geosearchef, 2019-07-21:
Ok, thx for confirming my suspicion. What do you mean by "this issue"? Have you done any investigation into the hw_data_upload flags/AR_CRCErr or does it just not work? I am getting a packet with a sounding flag from those routers.

A new router just arrived :), will test and keep you updated. If you're working on this for longer and are interested we can also share our knowledge via a different medium (email / instant messaging...).

b00l3an, 2019-07-22:
The issue I am referring to is the missing sounding packet. (I think) 
When I send from the client nothing happens on the receiver, however, if I leave the recv script running and have the client still attached I will get data received. I have yet to interpret the dat file so not sure what its collecting yet.

Geosearchef, 2019-07-22:
I have run tests with a modified CSI tool (in kernel space), all packets are received in the QCA9563, they have the not sounding flag not set, are therefore sounding. They get filtered here mostly https://github.com/xieyaxiongfly/Atheros-CSI-Tool/blob/master/drivers/net/wireless/ath/ath9k/ar9003_mac.c#L611 which tries to detect jumbograms consisting of many subframes and only wants the last subframe. According to datasheet this is only set if a CRCErr occurs.

The main issue though is that those flags https://github.com/xieyaxiongfly/Atheros-CSI-Tool/blob/master/drivers/net/wireless/ath/ath9k/ar9003_csi.c#L249 are never set which indicate the hardware (QCA9563) is currently uploading CSI data to the kernel driver.

huixiong73, 2019-10-04:
> This sounds like the same issue I have, although the packets are indeed received by the ath9k driver.
> 
> I have run tests with a modified CSI tool (in kernel space), all packets are received in the QCA9563, they have the not sounding flag not set, are therefore sounding. They get filtered here mostly https://github.com/xieyaxiongfly/Atheros-CSI-Tool/blob/master/drivers/net/wireless/ath/ath9k/ar9003_mac.c#L611 which tries to detect jumbograms consisting of many subframes and only wants the last subframe. According to datasheet this is only set if a CRCErr occurs. (I believe this to be a bug in the tool)
> 
> The main issue though is that those flags https://github.com/xieyaxiongfly/Atheros-CSI-Tool/blob/master/drivers/net/wireless/ath/ath9k/ar9003_csi.c#L249 are never set which indicate the hardware (QCA9563) is currently uploading CSI data to the kernel driver.

Thanks for sharing this information. I have a setup with TP-Link Ac1750 Archer c7 v2 and v4, v5 routers to receive CSI. The tool works fine if receiving CSI on v2 router but not with v4 and v5 routers. Sending works with all routers. Have you succeeded to make the CSI tool work on Archer C7 v5(QCA9563) by modifying the code?

Geosearchef, 2019-10-04:
I did not succeed. I assume the beamforming functionality is not present in newer Chips. The datasheets also don't mention TxBF as a feature anymore. I'm surprised the QCA9558 worked though. Setting the NOT SOUNDING flag seems to work as you are able to use them as transmitters.

Be aware though that unlike the v4/v5, the Archer c7 v1-v3 use 3 INTERNAL antennas on 2.4 GHz.

salmanpolito, 2020-01-09:
> Thanks for sharing this information. I have a setup with TP-Link Ac1750 Archer c7 v2 and v4, v5 routers to receive CSI. The tool works fine if receiving CSI on v2 router but not with v4 and v5 [...]

Hey.. I am trying to install CSI tool on archer C7-V2 router. I have applied the patch and have checked the code lines one by one. it can alse grabe ar9003_csi when logged in to the router, but when I try to send packets to the same other router. The other router cannot receive any packet (although it is in listening mode). Can you please help what shall I do. It will be grate if you could share your subupgrade bin file. many thanks