xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #39: Can anyone share captured CSI data?

b00l3an, 2019-07-29:
Can anyone share captured CSI data?
I had hoped to run my own experiments, but I have run out of time on my project. If anyone has the following scenarios captured using the CSI tool, I would be very grateful if you could share.

- 2 different devices in the same location; each captured separately
- 2 of the same devices in the same position; each obtained separately 

I was trying to determine if it was possible to distinguish device type based on their CSI signature.

Again if anyone can share the data for these scenarios, or anything similar even, I would be very grateful.

b00l3an

Geosearchef, 2019-07-29:
Wrote you an email.

b00l3an, 2019-07-29:
Thank you very Very much Geosearchef.

As soon as I have the thesis written up I will send you a copy.

putterer, 2020-06-09:
@shikhalakra22 
As part of my bachelor's thesis about Indoor Localization (https://www.researchgate.net/publication/342009154_Indoor_Localization_Using_Commercial_Off-The-Shelf_WiFi_Devices)
I have built my own user space server which reads the CSI from the kernel driver and sends it to subscribed clients via UDP. (can be found here: https://github.com/putterer/csi-server-openwrt, packet format in core.c)

Any client running in the network (https://github.com/putterer/csi-client-app) can then subscribe to the server and display, process and record CSI information in real time.

The following CSI was recorded in the format of my client app (DIFFERENT from the one used by the tool's user space app) which is a json serialialization of this class https://github.com/putterer/csi-client-app/blob/master/src/de/putterer/indloc/csi/CSIInfo.java.

https://drive.google.com/file/d/1qiS_3sLfhrCaeFRbZ17ikMAdeHNvsyDJ/view?usp=sharing
It contains multiple scenarios: the router in 2 different locations for 10 different angle of arrivals for 4 different distances (2-10 meters)

The entire folder can also be loaded as a recording and replayed in real time (for visualization/analysis/processing) using my tool (or just by reading the json e.g. in jupyter).

tweigel-dev, 2020-10-21:
@b00l3an and @shikhalakra22  I get CSI matrix values by the Intel5300 csi tool. Now iam trying to use Atheros. If i gather some data with atheros-csi, i will share them here.
@putterer  Nice and interesting thesis! 
maybe you can have a look at https://github.com/xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src/issues/55, your expertise could be very helpful here?