xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #40: Plot the Amplitude and Phase vs Time

magienguyen, 2019-10-21:
Is there anyone succeed to plot the amplitude and phase of the csi vs time? Please share. 
Many thanks

Geosearchef, 2019-10-21:
What exactly do you need? Multiple CSI packets captured over time? I can offer line of sight standing or moving csi with 30-500 packets (depending on the recording, will have to see what I have lying around). That's on 2.4GHz.

magienguyen, 2019-10-21:
I'm working on Wifi sensing using CSI. I got the CSI and trying to process the data by drawing graphs like Amplitude and Phase vs Subcarriers. For example, after getting 500 of packets, I use chanel 36 and 5GHz, so the CSI matrix is 3x3x56. I want to draw a graph of Amplitude and Phase of CSI elements with respect to time. To be specific, for each packets, I wil get the values of CSI in all 56 subcarriesr for example in position [1,1] of the matrix to draw a line. And then repeat that for 500 packets to get the graph of all elements [1,1]. Finally, I will get 9 graphs from 3x3 matrix. I used that to draw the graph of Amplitude and Phase with respect to subcariers index, now I want to do it with respect to time