xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #45: TP-Link Archer C7 v5 Firmware

fathanpranaya, 2019-12-10:
Hi, I followed this guideline https://github.com/xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src/wiki/Install-OpenWRT-version-of-Atheros-CSI-tool
and tried to build from source for TP-Link Archer C7 V5. 

I have successfully built the firmware, However, when I was trying to flash the built image to the router (using TFTP method) the router kept using the original TP-Link firmware.

Is there anyone here having the same issue?

Geosearchef, 2019-12-10:
Make sure to flash the factory and not sysupgrade image.
Also, sry for the bad news, but obtaining CSI on the ArcherC7v5 (QCA9563) is not possible. This is likely due to the newer QCA chips not performing hardware_upload for the TxBF functionality.

I had those devices as well. I now succeeded using the TP-Link WR2543ND (AR9380 on 2,4 and 5 GHz using 3 external antennas) and the WDR4300 (also called N750, AR9344 on 2,4 GHz with 2 external antennas, AR9580 on 5 GHz with 3 external antennas). See my issue for more info.

Geosearchef, 2019-12-10:
https://github.com/xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src/issues/35

fathanpranaya, 2019-12-11:
Thank you for quick response @Geosearchef 

anw, do you have any hardware recommendation for running the ubuntu version?

Thank you

Geosearchef, 2019-12-11:
The AR9462. It was a really popular chip which has been rebranded under the name AR5B22.

fathanpranaya, 2019-12-12:
@Geosearchef again.. thanks a lot!

last, you told me that you also tried TP-Link ArcherC7v5. is it possible to share the OpenWRT firmware you built for ArcherC7v5?

Geosearchef, 2019-12-12:
Use this at your own risk. It might also contain some debug statements in dmesg from the CSI tool kernel code.

https://files.geosearchef.de/index.php/s/AkhG0eckoyUaEY9

Beware though, as far as I could see, the hardware_upload flag from thr chip specified in the datasheet is never set, othrr people had no success either, so this is likely not going to be able to obtain csi. (but you can use the A c7v5 to transmit and receive on another device, setting the sounding flag works)

fathanpranaya, 2019-12-13:
Thanks a lot dude @Geosearchef ~

closed 2019-12-13 by fathanpranaya