xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #48: 4.1.10+ kernel doesn't run wifi on Ubuntu 16.04, even though it was working with the kernel it came with.

vibhorrawal, 2020-02-21:
The specification of laptop is:
Make: Asus
Model: X556UQK
Processor: Intel® Core™ i5-7200U CPU @ 2.50GHz × 4 
Wi-Fi NIC: Qualcomm Atheros QCA9377

I had flashed with Ubuntu 16.04 amd64 (kernel: 4.10.0-28-generic), wifi was working, but after following the steps to install the modified kernel (which was successful) I am not able to make the Wi-Fi work. We need to use the Atheros CSI logging tool, if there is some workaround, please let me know.

Geosearchef, 2020-02-21:
I didn't succeed on Ubuntu 16.04 either. The Atheros CSI Tool is built for 14.04. Try installing that.

s10singh97, 2020-02-21:
I think the issue is with the kernel and not with the Ubuntu version. I am also stuck here. Any lead on this would be very helpful.

Thanks

Geosearchef, 2020-02-21:
Why not just use Ubuntu 14.04? You can try building and debugging the kernel yourself (or actually porting the tool to a newer kernel version).

I still suspect the newer Ubuntu version doesn't like that really old kernel.

vibhorrawal, 2020-02-22:
Tried 14.04, neither was there native wi-fi support (with kernel out of box), nor after successful installation of 4.4.10+ could I make the wi-fi work

1037913794, 2020-03-22:
I succeed, you should reinstall your firmware.