xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #51: Using QCA9880 WiFi NIC with ath9k through backward compatibility support

darkreavo, 2020-04-06:
Dear all,

I was wondering if a QCA9880 powered WiFi NIC such as the Compex WLE900VX can be used by the Atheros CSI tool with the ath9k driver through backward compatibility support. The reason why I am asking this is that I want to use the Atheros CSI tool in a measurement setup consisting of mini PC's featuring a WLE900VX WiFi NIC. Unfortunately AR9580/9590 WiFi NIC's such as the Compex WLE350NX are not distributed anymore.

The Compex website states that the Compex WLE900VX is backward compatible with 802.11 a/b/g/n/ac. The Atheros CSI tool hardware webpage features a Compex WLE900VX image while link2 provides a different kind of NIC. 

Thank you in advance for your answer.

Regards,
darkreavo

csmibrahim, 2020-10-25:
Hi darkreavo,
Did this work for you?

Aleksanaa, 2023-09-01:
Any progress?