xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #55: Gather CSI values at Rx  with only at one device (rx); NDP sounding

tweigel-dev, 2020-10-21:
Hay, 
the Atheros CSI Tools is a great and fascinating project. 

I played around with the Intel5300 CSI tool by halperin. His tool gathers csi values by the Rx without installing the tool at the Tx. It works nice but with different amount of data according to the chip of the Tx device. 
Now i want to try the Atheros tool.

I my opinion the main feature difference is that :
- atheros supports more chips as Rx
- intel csi work with installed tool only at Rx (hardware space limitations after driver modification)
- atheros csi supports WPA/WPA2 encryption

Now i want a system that gathers csi information of WPA2 encrypted networks without installing software at the Tx.

### My Questions:
Is there a configuration or implementation to get CSI matrix with the atheros as Rx without installing something at the Tx?

I read that the changes which are done at this tool is to set the sounding flag, to send an NDP before the main datagram. But in my research i found out, that NDP sounding are invented for 802.11 ac. but the atheros chips are 802.11 n. Where is the correlation between them?

putterer, 2020-10-21:
>I played around with the Intel5300 CSI tool by halperin. His tool gathers csi values by the Rx without installing the tool at the Tx.

Wow, I wasn't able to get that running with the receiver using the IWL5300 and the transmitter being a WDR4300 with an AR9580, even when playing around with the injector / inspecting the raw wifi traffic.

>I my opinion the main feature difference is that :
 - the atheros tool returns all 54 / 114 instead of just 30 subcarriers
 - the atheros tool uses 10 instead of 8 bit precision, I didn't test yet how precise the actual data is though
 - The amplitude data from the atheros chips is not stable over time compared to the IWL5300 based data

> Now i want a system that gathers csi information of WPA2 encrypted networks without installing software at the Tx.
Take a look at Nexmon. I think they are capable of doing that, their firmware runs on some specific broadcomm chips. They used to support only the Nexus 5 phone but recently added support for some Galaxy devices as well as the RaspPi B3+/B4 which should be quite interesting.

> Is there a configuration or implementation to get CSI matrix with the atheros as Rx without installing something at the Tx?
No. As far as I know there is not. Take a look at a datasheet of any of the Atheros chips. (e.g. https://datasheetspdf.com/pdf-file/825113/Atheros/AR9344/1)
Page 102:   according to the 802.11n specification, CSI / a steering matrix can only be calculated when the PPDU is sounding. In my tests I could never get the not_sounding field to not be set without changing it's value on the TX chip being an Atheros chip. The hw_upload_data_valid and hw_upload_data will indicate to the kernel when CSI or a steering matrix has been uploaded from the firmware to the kernel.

As far as I know it should be possible to use any device as the transmitter as long as the sounding flag is set. This might be possible without atheros devices if you manage to modify any other kernel driver, e.g. of the broadcomm chips (which are present in most smartphones) to set it.

>  But in my research i found out, that NDP sounding are invented for 802.11 ac. but the atheros chips are 802.11 n.
NDP souding is specified in 802.11n, although in the case of 802.11n, different manufacturers implemented different features subsets of beamforming while in 802.11ac the standard required it for interoperability.

Take a look at the 802.11n standard (https://ieeexplore.ieee.org/document/5307322)
 - chapter "9.17 Sounding PPDUs" (page 161),
 - chapter "9.21 Null data packet (NDP) sounding" (page 181)
 - for a general overview of TxBF:   "9.19 Transmit beamforming" (page 164)

About the IWL5300 using implicit beamforming, I'm not sure, I thought both of them used explicit TxBF. From my thesis:
--------
4.3.1  Transmit Beamforming
The 802.11n amendment to the WiFi standard [1] defines procedures for transmit beamforming (TxBF) to increase the range and signal strength of WiFi networks. The goal is to calculate a steering matrix based on the received Channel State Information at the client which can then be used by the AP to steer the transmitted signal in the direction of the client.There are two types of beamforming defined in 802.11n, explicit and implicit feedback. In implicit mode the channel is estimated by the beamformer (AP) over a longer time, in explicit mode the beamformer sends an explicit request (a training sequence) to the beamformee which then calculates the Channel State Information or steering matrix and transmits it back to the beamformer [1, 9.19.1].  Both tools listed in sections 4.4 and 4.5 use explicit beamforming.To obtain CSI, the beamformee needs to receive either a sounding PLCP Protocol DataUnit (PPDU) or Null Data Packet (NDP), which means the sounding parameter has to be set by the transmitter [1, 9.17] which restricts the usage of CSI based applications with non-modified WiFi devices. Implementations of TxBF in 802.11n chips are mostly not present or not fully complete as the standard is not very specific and offers too many options (implicit and explicit beam-forming) which prompted many manufacturers to implement this feature only in part. This leads to some problems when operating different devices from different manufacturers. In case of the Atheros ath9k driver e.g., CSI reporting is not even implemented. While the chip does upload the calculated CSI matrix to software, it will not be reported back to the beamformer [13].
--------
(https://www.researchgate.net/publication/342009154_Indoor_Localization_Using_Commercial_Off-The-Shelf_WiFi_Devices)

tweigel-dev, 2020-10-21:
Wow thanks for the fast and detailed response!
### intel csi
if you want to use the intel csi. spanev updated the kernel and make it working for a explicit ubuntu 18.04 version.
https://github.com/spanev/linux-80211n-csitool 
and have a look to this nice MR for python usage if you hate matlab like me ^^:
https://github.com/pgawlowicz/linux-80211n-csitool-supplementary 
### nexmon-csi
I'am watching the progress of nexmon-csi for a few weeks and its really interesting! But currently there is no chip with 3 or more antenna.
### esp-csi
https://github.com/jonathanmuller/ESP32-gather-channel-state-information-CSI-
here is also another csi tool for the esp chipset. But here the same problem. No chip with 3 or more antenna
### 802.11n/ac
I interpreted [this](https://www.oreilly.com/library/view/80211ac-a-survival/9781449357702/ch04.html) at  chapter 4 that TxBF  get inserted at 802.11ac but yours seems to be right.

mzakharocsc, 2020-10-21:
nexmon-csi has 4 antenna router supported. what do you mean no chip with 3 or more antenna?

tweigel-dev, 2020-10-21:
But if Intel-csi and atheros-csi are both using explicit-beamforming why intel-csi don't need an implementation at the tx?
interesting...

putterer, 2020-10-21:
>But if Intel-csi and atheros-csi are both using explicit-beamforming why intel-csi don't need an implementation at the tx?

Because it's implemented in the Atheros firmware which we have no access to while the authors of the Intel tool had access to the firmware of the IWL5300 and modified it when developing their tool.

> and have a look to this nice MR for python usage if you hate matlab like me ^^:

Ooof, that would have been really nice to know for investigating the Intel csi format.
I do not like python either. (it's better than matlab and nice for maths, but still, dynamic typing, meh)
Here's my C implementation of the C and matlab mixed processing of the tool: https://gist.github.com/putterer/19ab5c4de86a877985654cec51d14468 

I didn't succeed at recompiling the user space app so I opted for printing the log_to_file output to stdout and piping that into my csi-server which forwards it to my client, so I can preview and process intel and atheros csi in real time simultaneously.

> here is also another csi tool for the esp chipset. But here the same problem. No chip with 3 or more antenna

I also considered that, but needed more than one antenna for Indoor Localization.

tweigel-dev, 2020-10-21:
@mzakharocsc 
hay, thanks for the response.
at [ASUS_RT-AC86U](http://en.techinfodepot.shoutwiki.com/wiki/ASUS_RT-AC86U) is written that there is  this chip: Broadcom BCM4365E or Broadcom BCM4366E
interesting is that i cant find any useful mention of the bcm4366c0
is the "c0" and the "E" an appendix? sry iam not that in at hardware version-tagging.
is it possible that the nexmon-csi is compatible or expandable with 
[ASUS_GT-AC5300](http://en.techinfodepot.shoutwiki.com/wiki/ASUS_GT-AC5300)?