xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #56: compatibility of the tool on tp-link wdr4900

csmibrahim, 2020-10-28:
Can I install the tool on TP-Link TL-WDR4900 (N900) v1.3?

https://openwrt.org/toh/tp-link/tl-wdr4900

csmibrahim, 2020-10-30:
Also is TP-Link TL-WR841N, Version:11.1 supported by the tool?

https://openwrt.org/toh/tp-link/tl-wr841nd

csmibrahim, 2020-10-30:
I do not know why I find it hard to check for the router compatibility  with the tool? 

Like for example when I open the router page on OpenWrt website: what I have to check exactly? 

Are only those routers mentioned on this list (https://github.com/xieyaxiongfly/OpenWRT_firmware) supported by the tool?

Kindly give help me.

Geosearchef, 2021-01-15:
It's the openwrt firmware patched with the kernel driver modification for the CSI tool.

Geosearchef, 2021-01-15:
The WDR4900 should work using the 5GHz AR9580 chip, using QCA9558 (2,4GHz) with the tool might or might not work. (see https://github.com/xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src/issues/35)

closed 2021-07-29 by csmibrahim