xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #59: ./send_Data: 1: ./send_Data: syntax error: "(" unexpected

serrhini, 2021-01-15:
Hi all
the receive mode is working, however sendData is not working I got the following error after executing senData with wlan0 and mac addres of destination and 100 
error message  ./send_Data: 1: ./send_Data: syntax error: "(" unexpected
please can you enlighten me

miligithub, 2021-03-07:
I met with the same problem. After spending a whole day on this problem, I found out that I should have use the `/bin/sendData` instead of `scp` the `send_Data` file from [the UserSpaceApp folder](https://github.com/xieyaxiongfly/Atheros_CSI_tool_OpenWRT_UserSpaceApp_src/tree/master/sendData).

kr689, 2023-01-25:
Why is this the case? I'm trying to use a modified version of sendData and so I used scp but I'm getting this error. The normal sendData works fine. Did you manage to fix it?

[image]
See the last lines of the terminal