xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #60: Matlab crashes when reading CSI

shazal, 2021-05-11:
Hey I have been able to capture CSI successfully but when I try to read it, Matlab crashes. Any idea why it might be happening and what steps I should take to fix it? Thank you.

Geosearchef, 2021-05-11:
What are you doing? Which code did you use?
Do you get an error? What do you mean by crash? The entire IDE just disappears?
In that case you should take a look at the log files / ask in a MATLAB forum / support platform.

Geosearchef, 2021-05-11:
Might be related to you running on OS X. The error seems to be located in the native code the csi tool uses for unpacking the data from the recording: https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/blob/master/matlab/read_csi.c

It might also be that your recording is corrupt.

CSIhicham, 2022-01-23:
Hello, 
I have the same problem,  What should I do to fix it?

Thanks