xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #62: csi length 0

fpeci, 2021-06-20:
Two devices with openwrt
1. tplink 841Nv11 - Qualcomm Atheros QCA9533 (able to send and receive)
2. Archer c7 v1.1 - Qualcomm Atheros QCA9558 (able to send and receive)
lorcon package doesnt have option to specify modulation index on both APs
1st AP is setup in AP role
2nd AP is setup as client

trying to read the file with Matlab script shows no csi values (although it is filled with rssi values) also csikit doesn't recognize it as Atheros file .dat file that I am trying to read after running recvCSI command

are these two hardware boxes supported ?

Cheers

Geosearchef, 2021-06-20:
QCA9558 should work. If you habe RSSI, the issue is probably not with the devices, as they are running the same software. You should probably take a look at the data.

fpeci, 2021-06-20:
were you able to use the injector, it is not working for me, doesn't have the same options as in the wiki ?

h0r4ci0, 2024-03-24:
Hi, I have the same issue, have you solved it? I'm a bit lost 😅

After receiving HT packets and analyzing the data in Matlab, the values for the variable ‘csi’ are consistently zero.

I am using two laptops with Ubuntu:

The sender laptop runs Ubuntu 22.04.4 LTS.
The receiver laptop runs Ubuntu 14.04 LTS with kernel 4.1.10+ and a Qualcomm Atheros AR9462 network card.

[image]

[image]

Geosearchef, 2024-03-24:
Is your sender setting the sounding flag? (or setting the not_sounding to false)

You need to set the flag for the receiver chip to calculate and report CSI, if I remember correctly.

The 14.04 modified driver does that.

h0r4ci0, 2024-03-25:
I think you're right, I'm not sending the sounding flag, or at least that's what I understand when analyzing it with Wireshark.

Regarding the modified driver, do you mean the Ubuntu 14.04 version with kernel 4.1.10+? I haven't used that version on the transmitter device due to issues with the network interfaces, as it didn't recognize any.

Should I try to get that specific version working, or can I change some settings from the one I'm currently using (22.04.4 LTS) to make it functional? Thanks!

[image]

Geosearchef, 2024-03-25:
It's been some time, but if memory serves, I modified the ath9k driver to set the sounding flag. Just search this repository for the not_sounding and look at the Git history.
You would have to modify the driver you are using, assuming your hardware supports that option.

It might be easier to build and use the old kernel (4.1.10+) on Ubuntu 14.04 though, if your Hardware is compatible with the modified ath9k driver here.

h0r4ci0, 2024-03-26:
Okay, thank you very much! I'll be testing it out over the next few days. I'll post any progress here.