xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #63: make: *** [world] Error 1

Precela, 2021-11-08:
Hi,
I am new with OpenWRT. I am actually trying to install Atheros_OpenWRT csi tool. But i am having trouble with installation. I am using Ubuntu 18.04 and trying to install it on TL Archer C7 router. 
This is currently on WSL but i have tried it on Ubuntu VM and I have also installed Ubuntu OS in raspberry pi and tried that too. I get the same error everywhere. 

When i command  

> ./scripts/feeds install -a

This is the result i get:

root@DESKTOP-NMFVA7A:/home/newhome/Atheros_CSI_tool_OpenWRT_src# ./scripts/feeds install -a
Collecting package info: done
Collecting target info: done
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libidn2', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libssh2', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/lorcon/Makefile' has a build dependency on '+libpcap', which does not exist
WARNING: Makefile 'package/lorcon/Makefile' has a build dependency on '+libnl', which does not exist
WARNING: Makefile 'package/lorcon/Makefile' has a build dependency on '+wireless-tools', which does not exist
WARNING: Makefile 'package/lorcon/Makefile' has a build dependency on '+libnfnetlink', which does not exist
feeds/packages/net/gnunet/Makefile: invalid group spec :postgres=5432
Installing all packages from feed packages.
feeds/packages/net/gnunet/Makefile: invalid group spec :postgres=5432
Installing package 'asterisk' from telephony
WARNING: No feed for source package 'libxml2' found
WARNING: No feed for package 'jansson' found
WARNING: No feed for package 'libcap' found
WARNING: No feed for package 'libedit' found
WARNING: No feed for package 'libsqlite3' found
WARNING: No feed for package 'libxml2' found
WARNING: No feed for package 'mpg123' found
WARNING: No feed for package 'uw-imap' found
WARNING: No feed for package 'unixodbc' found
WARNING: No feed for package 'libsqlite3' found
WARNING: No feed for package 'alsa-lib' found
WARNING: No feed for package 'portaudio' found
Installing package 'dahdi-tools' from telephony
Installing package 'dahdi-linux' from telephony
Installing package 'libpri' from telephony
WARNING: No feed for package 'bluez-libs' found
WARNING: No feed for package 'libspeex' found
WARNING: No feed for package 'libspeexdsp' found
WARNING: No feed for package 'libogg' found
WARNING: No feed for package 'libspeex' found
WARNING: No feed for package 'libvorbis' found
WARNING: No feed for package 'libspeex' found
WARNING: No feed for package 'libspeexdsp' found
WARNING: No feed for package 'unixodbc' found
WARNING: No feed for package 'libpq' found
Installing package 'pjproject' from telephony
Installing package 'libsrtp' from telephony
WARNING: No feed for package 'libical' found
WARNING: No feed for package 'libneon' found
WARNING: No feed for package 'libneon' found
WARNING: No feed for package 'libical' found
Installing package 'iksemel' from telephony
WARNING: No feed for package 'libgnutls' found
WARNING: No feed for package 'libneon' found
WARNING: No feed for package 'libical' found
WARNING: No feed for package 'libneon' found
WARNING: No feed for package 'libopenldap' found
WARNING: No feed for package 'libmysqlclient' found
Installing package 'spandsp' from telephony
WARNING: No feed for package 'libtiff' found
WARNING: No feed for package 'libtiff' found
[... 2008 lines cut ...]
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking whether _XOPEN_SOURCE should be defined... no
checking for Minix Amsterdam compiler... no
checking for x86_64-pc-linux-gnu-ar... no
checking for ar... ar
checking for _LARGEFILE_SOURCE value needed for large files... no
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... no
checking for x86_64-pc-linux-gnu-ar... ar
checking for special C compiler options needed for large files... (cached) no
checking for _FILE_OFFSET_BITS value needed for large files... (cached) no
checking for inline... inline
che

Geosearchef, 2021-11-08:
According to your log:

```
error: in /home/newhome/Atheros_CSI_tool_OpenWRT_src/build_dir/host/tar-1.30': configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check) See config.log' for more details
```

Which version of C7 are you trying to install the tool on?

Precela, 2021-11-08:
@Geosearchef  I am trying to install Version 1 of C7.

Geosearchef, 2021-11-08:
There's a prebuilt image for C7 v1-3.

Precela, 2021-11-08:
I had no idea. Where can i find it? or is it already in the router?

Geosearchef, 2021-11-08:
https://github.com/xieyaxiongfly/OpenWRT_firmware

Precela, 2021-11-08:
@Geosearchef  Thank you so much. So i can just flash it on the router then?

Geosearchef, 2021-11-08:
You should be able to, unless you want to modify the kernel. If you want to modify the recv_csi tool though, e.g. to ship off the obtained CSI data via UDP in real time (https://github.com/putterer/csi-server-openwrt), you'll need a toolchain for cross compiling which you can only obtain by compiling the image yourself.

Precela, 2021-11-08:
Thank you so much for all your help.
But i still don't understand what i did wrong. Do you know what was wrong
with the way i was building the image? Should i have done something
differently?

Geosearchef, 2021-11-08:
You're running the command from the root user. Run it again (probably after clearing up / deleteing and recloning the repo) as a normal user. Alternatively, set the flag mentioned in the error at the end.

```
FORCE_UNSAFE_CONFIGURE=1 ./scripts/feeds install -a
```

closed 2022-03-30 by Precela