xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #64: Minimal setup for running CSI extraction

Itamar-Gozlan, 2021-12-05:
I wonder what the minimum requirement is in terms of devices before using this CSI tool.
Up until now, I have installed the CSI tool on TP-LINK WDR4300 using OpenWRT.
Do I have to install this tool on another device (Ubuntu/OpenWRT) to make it work? 

Following the Atheros CSI Tool (OpenWrt) - User Guide, the examples introduce two devices (Device A and Device B) but do not specify the router, and can they both be routers/PCs?

Can I use WSL (Linux on top of Windows) to install and use the CSI tool, or do I have to install ubuntu OS directly?

Geosearchef, 2021-12-08:
You will require a supported device running the CSI tool as receiver (e.g. the WDR4300 you have) and one that sets the sounding flag as transmitter. The CSI tool does this automatically, as far as I can see also on devices that are not supported as receiver (e.g. the AR9563 in the Archer C7v5).

You will likely need another atheros chip running the CSI tool. This can be a conventional Router or an x86 device.
I'm pretty certain, that you won't be able to use WSL, as you won't be able to modify the wireless driver there (as it's controlled by the windows kernel). You will also likely need a special version of the Linux Kernel (~4.2, not sure atm, I suggest running Ubuntu 14.04 and downgrading the kernel if needed).

Itamar-Gozlan, 2021-12-08:
Thank you very much for the detailed answer

closed 2021-12-08 by Itamar-Gozlan