xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #65: fatal:invalid revision range during updating feeds

ng91, 2022-01-15:
I'm getting following errors when trying to update feeds:

> Updating feed 'packages' from 'https://git.openwrt.org/feed/packages.git' ...
Cloning into './feeds/packages'...
remote: Enumerating objects: 6472, done.
remote: Counting objects: 100% (6472/6472), done.
remote: Compressing objects: 100% (5388/5388), done.
remote: Total 6472 (delta 212), reused 4927 (delta 138), pack-reused 0
Receiving objects: 100% (6472/6472), 4.28 MiB | 3.48 MiB/s, done.
Resolving deltas: 100% (212/212), done.
Create index file './feeds/packages.index' 
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'time'... ok.
Checking 'perl'... ok.
Checking 'python'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'ldconfig-stub'... ok.
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
ERROR: please fix feeds/packages/devel/pkgconf/Makefile - see logs/feeds/packages/devel/pkgconf/dump.txt for details
ERROR: please fix feeds/packages/libs/freetype/Makefile - see logs/feeds/packages/libs/freetype/dump.txt for details
ERROR: please fix feeds/packages/libs/glib2/Makefile - see logs/feeds/packages/libs/glib2/dump.txt for details
ERROR: please fix feeds/packages/libs/jose/Makefile - see logs/feeds/packages/libs/jose/dump.txt for details
ERROR: please fix feeds/packages/libs/json-glib/Makefile - see logs/feeds/packages/libs/json-glib/dump.txt for details
ERROR: please fix feeds/packages/libs/jsoncpp/Makefile - see logs/feeds/packages/libs/jsoncpp/dump.txt for details
ERROR: please fix feeds/packages/libs/libdrm/Makefile - see logs/feeds/packages/libs/libdrm/dump.txt for details
ERROR: please fix feeds/packages/libs/libesmtp/Makefile - see logs/feeds/packages/libs/libesmtp/dump.txt for details
ERROR: please fix feeds/packages/libs/libevdev/Makefile - see logs/feeds/packages/libs/libevdev/dump.txt for details
ERROR: please fix feeds/packages/libs/libinput/Makefile - see logs/feeds/packages/libs/libinput/dump.txt for details
ERROR: please fix feeds/packages/libs/liblz4/Makefile - see logs/feeds/packages/libs/liblz4/dump.txt for details
ERROR: please fix feeds/packages/libs/libmpdclient/Makefile - see logs/feeds/packages/libs/libmpdclient/dump.txt for details
ERROR: please fix feeds/packages/libs/libpsl/Makefile - see logs/feeds/packages/libs/libpsl/dump.txt for details
ERROR: please fix feeds/packages/libs/libslirp/Makefile - see logs/feeds/packages/libs/libslirp/dump.txt for details
ERROR: please fix feeds/packages/libs/libsoup/Makefile - see logs/feeds/packages/libs/libsoup/dump.txt for details
ERROR: please fix feeds/packages/libs/nlohmannjson/Makefile - see logs/feeds/packages/libs/nlohmannjson/dump.txt for details
ERROR: please fix feeds/packages/libs/p11-kit/Makefile - see logs/feeds/packages/libs/p11-kit/dump.txt for details
ERROR: please fix feeds/packages/libs/pixman/Makefile - see logs/feeds/packages/libs/pixman/dump.txt for details
ERROR: please fix feeds/packages/libs/spice/Makefile - see logs/feeds/packages/libs/spice/dump.txt for details
ERROR: please fix feeds/packages/libs/spice-protocol/Makefile - see logs/feeds/packages/libs/spice-protocol/dump.txt for details
ERROR: please fix feeds/packages/multimedia/grilo/Makefile - see logs/feeds/packages/multimedia/grilo/dump.txt for details
ERROR: please fix feeds/packages/multimedia/grilo-plugins/Makefile - see logs/feeds/packages/multimedia/grilo-plugins/dump.txt for details
ERROR: please fix feeds/packages/multimedia/gst1-libav/Makefile - see logs/feeds/packages/multimedia/gst1-libav/dump.txt for details
ERROR: please fix feeds/packages/multimedia/gst1-plugins-bad/Makefile - see logs/feeds/packages/multimedia/gst1-plugins-bad/dump.txt for details
ERROR: please fix feeds/packages/multimedia/gst1-plugins-base/Makefile - see logs/feeds/packages/multimedia/gst1-plugins-base/dump.txt for details
ERROR: please fix feeds/packages/multimedia/gst1-plugins-good/Makefile - see logs/feeds/packages/multimedia/gst1-plugins-good/dump.txt for details
[... 50 lines cut ...]
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
Collecting package info: done
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
Collecting target info: done
Updating feed 'telephony' from 'https://git.openwrt.org/feed/telephony.git' ...
Cloning into './feeds/telephony'...
remote: Enumerating objects: 291, done.
remote: Counting objects: 100% (291/291), done.
remote: Compressing objects: 100% (278/278), done.
remote: Total 291 (delta 9), reused 72 (delta 1), pack-reused 0
Receiving objects: 100% (291/291), 192.64 KiB | 3.32 MiB/s, done.
Resolving deltas: 100% (9/9), done.
Create index file './feeds/telephony.index' 
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
Collecting package info: done
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a7830185a86b8409bea57053222b738563abb6c9
Collecting target info: done

sarashoouri, 2024-06-07:
have you fixed it? I am getting the exact error.

ng91, 2024-06-08:
Yes, i found another repository - https://github.com/OctHe/atheros_csi_tool_openwrt

closed 2024-06-08 by ng91