xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #67: How to change the number of transmitting and receiving antennas? (TP-Link WDR4300)

AlexandrAstafiev, 2022-02-24:
Hello!

I purchased 2 routers and installed a modified openWRT based firmware on them. The number of antennas on the routers is 3. However, I receive 2x2x56 data packets. How can I change the number of antennas in packets? What needs to be changed? File on Matlab? sendData or recvCSI? Is there any instruction?

Geosearchef, 2022-02-24:
What routers? I'm guessing WDR4300? That one uses the two outer antennas with the AR9344. For 3 antennas you'll have to use the AR9580. (switching from 2.4 GHz band to 5 GHz)

AlexandrAstafiev, 2022-02-24:
Thanks for the quick response. Then what models can you recommend for receiving data from 3 or more antennas. It is desirable to obtain information from 114 subcarriers or more.

AlexandrAstafiev, 2022-02-24:
That is, the WDR4300 router can be switched to 5GHz and receive data from 3 antennas?

Geosearchef, 2022-02-24:
Yes

closed 2022-02-24 by AlexandrAstafiev