xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src #68: Receiver openwrt router not getting any packets

mamdouhJ, 2022-03-03:
Im using two openwrt wdr-4300 routers. Im using one as a transmitter and the other as a receiver. I created an AP on the transmitter and associated the receiver with the created AP. I connected the two routers with an ethernet cable to be able to ssh to them using 1 computer. When I run sendData from my transmitter to my receiver nothing is happening on the receiver side. I'm running recvCSI on the receiver router. 
Any help would be appreciated. Thank you

putterer, 2022-03-03:
Can you ping the receiver from the transmitter and get a response?
Are the ethernet and wifi networks using the same subnet?

mamdouhJ, 2022-03-03:
turns out that when I restart the transmitter the AP doesn't automatically start. That was the problem

closed 2022-03-03 by mamdouhJ